qilimanjaro-tech/qililab #570: [BUG] bug report

visagim, 2023-10-20:
### Expected behavior

Running attached code should return a 2*2 array

### Actual behavior

test

### Source code

```Shell
running

import os
import numpy as np
import qililab as ql
from qililab.typings import Parameter
from qililab.utils import Loop
from qibo.models import Circuit
from qibo.gates import X,Y,CZ,M
import matplotlib.pyplot as plt
from qililab.utils.qibo_gates import Wait
from qililab import Drag

ql.logger.setLevel(40)

platform_name = "/home/vsanchez/LabScripts/QuantumPainHackathon/runcards/soprano_master_galadriel_main.yml"

platform = ql.build_platform(runcard=platform_name)

platform.connect()
platform.turn_on_instruments()
platform.initial_setup()

circuit0 = Circuit(5)

circuit1 = Circuit(5)
circuit1.add(Drag(2,np.pi,0))
circuit1.add(CZ(2,4))
circuit1.add(M(2,4))

results0 = []
total = 0
niters = 1
num_bins = 10
for n in range(0, niters):
    result = platform.execute(program=circuit1, num_avg=1, repetition_duration=20, num_bins=num_bins)

results0.append(result)
circuit0 = Circuit(5)

circuit1 = Circuit(5)
circuit1.add(Drag(2,np.pi,0))
circuit1.add(CZ(2,3))
circuit1.add(M(2,3))

results = []
total = 0
niters = 1
num_bins = 10
for n in range(0, niters):
    result = platform.execute(program=circuit1, num_avg=1, repetition_duration=20, num_bins=num_bins)

results.append(result)
platform.disconnect()

print("results0", results0[0].array)
print("\nresults", results[0].array)
```

### Existing GitHub issues

- [X] I have searched existing GitHub issues to make sure the issue does not already exist.

visagim, 2023-10-23:
- [ ] Apply fix
- [ ] Check that fix solves problem

AlbertMitjans, 2023-10-27:
One thing we definitely need to do is clear the cache between executions. This line should be added inside `Platform.execute`:
[image]

closed 2023-10-30 by visagim