qilimanjaro-tech/qililab #580: qililab function to turn the data produced by qililab directly into the format received through qiboconnection

anapalu, 2023-10-26:
### Feature details

Maybe it would be outdated fast (up to you to judge), but it would be useful to have a fast conversion between formats to debug one channel of doing the experiment vs the other

### Implementation

I assume this should just be creating a method that implements the same conversion that is automatically applied through qiboconnection.

### How important would you say this feature is?

1: Not important. Would be nice to have.

AlbertMitjans, 2023-10-26:
Thanks for opening this Issue @anapalu! We were thinking of changing `qiboconnection` instead of `qililab`. But I completely agree, both libraries should return the same format.

Having said that, the cloud team currently has no capacity to make improvements to the code, thus this will have to wait...

visagim, 2023-10-27:
Hi, the results output from qiboconnection e.g.:
```python
>>> job_id = connection.execute(circuit=circuits, nshots=2000)
>>> results = connection.get_result(job_id=job_id[0])
>>> results
[{'probabilities': {'00000': 0.192,
   '00001': 0.212,
   '00010': 0.0295,
   '00011': 0.029,
   '00100': 0.0595,
   '00101': 0.2305,
   '00110': 0.007,
   '00111': 0.0365,
   '01000': 0.0275,
   ...
```
should be equivalent to the `results.probabilities() method within qililab, since qiboconnection is running this in the `execute_circuit` method:
```python
  connection = cls._create_connection()
  remote_runcard = connection.get_runcard(runcard_name=PLATFORM_NAME)
  # pylint: disable=no-member
  results = ql.execute(
      program=circuits, runcard=remote_runcard.runcard, nshots=nshots
  )  # pylint: disable=no-member
  if not isinstance(results, list):
      results = [results]
  return [{"probabilities": result.probabilities()} for result in results]
```
where the `result` object above is the same `result` object you obtain when in qililab you run something of the sort:
```python
  result = platform.execute(program=my_chsh_circuit_transpiled,
                            num_avg=HW_AVG,
                            repetition_duration=REPETITION_DURATION,
                            num_bins=NUM_SHOTS)
```

Does this solve your issue?

anapalu, 2023-10-27:
Yup, thanks!

closed 2023-10-27 by anapalu