qilimanjaro-tech/qililab #582: [BUG] `Asserts` inside `with` are not really asserting anything in qililab tests.

GuillermoAbadLopez, 2023-10-26:
### Expected behavior

We should assert things

### Actual behavior

Asserts inside `with`, never make a test fail! 
[image]

### Existing GitHub issues

- [X] I have searched existing GitHub issues to make sure the issue does not already exist.

GuillermoAbadLopez, 2023-10-26:
What acuatlly is happening :

The error makes you leave the with, so the assert is never actually run
(I just debugged this, and its true, you never arrive to the assert, since the erro makes you leave the with)

AlbertMitjans, 2023-10-26:
To be more precise, this only happens when using `with pytest.raises`.

GuillermoAbadLopez, 2023-10-26:
There was only one test doing this incorrectly in the end, and has been changed! 
Clossing the Issue 👍

closed 2023-10-26 by GuillermoAbadLopez