qilimanjaro-tech/qililab #638: [BUG] Qblox blow up if I cancel the measurement

TronGon, 2023-12-11:
### Expected behavior

I expect to be able to cancel a measurement in the fly and the thing to don't stop working.

### Actual behavior

Turns out that I realised that the array for the coil was too long, so I cancelled the operation. Now I get an error if I lunch a new measurement. If I reboot the platform, I wil loose all the parameters as they are now, which will mess with the experiment, this cannot happen.

### Additional information

I want to do a flux measurement vs frequency. Currently I am using Yokogawa as current source, RS as the microwave generator and the Qblox to read the signal. 

### Source code

```Shell
lor_freq_array = np.linspace(6.83e9,6.86e9,21)
coil_current_array = np.linspace(0e-3,5.4e-3,51)
# coil_current_array = np.linspace(0.5e-3,1e-3,3)

platform.set_parameter(alias=f'M({QUBIT})', parameter=ql.Parameter.AMPLITUDE, value=0.5)
aux = []
M = []
M_sin_complejos = []

meas =  qc.Measurement(exp=exp, name='ramping_up_current')
meas.register_parameter(f)
meas.register_parameter(cc)
meas.register_parameter(iq, setpoints = (f,cc))

with meas.run() as datasaver:

    for cur in tqdm(coil_current_array, colour='cyan'):
        
        platform.set_parameter(alias=f'flux_q{QUBIT}_bus', parameter=ql.Parameter.CURRENT, value=cur, channel_id=0)

        for freq in lor_freq_array:
            platform.set_parameter(alias='readout_bus', parameter=ql.Parameter.LO_FREQUENCY, value=freq, channel_id=0) # , channel_id=QUBIT_IDX), , channel_id=0
            result = platform.execute(program=circuit, num_avg=HW_AVG, repetition_duration=REPETITION_DURATION)
            aux.append(result.array[0] + result.array[0]*1j)  

            f(freq)
            cc(cur)
            iq(result.array[0][0] + result.array[1][0]*1j)

            datasaver.add_result((f, f()),
                                 (cc,cc()),
                                    (iq,iq()))
            dataset = datasaver.dataset

plot_dataset(dataset)
```

### Tracebacks

```Shell
An exception occured in measurement with guid: 2d002c65-0000-0000-0000-018c581ee365;
Traceback:
Traceback (most recent call last):
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/scpi/cluster.py", line 50, in decorator_wrapper
    return func(*args, **kwargs)
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/scpi/cluster.py", line 2726, in _get_sequencer_config
    var0 = self._read_bin('SLOT{}:SEQuencer{}:CONFiguration?'.format(slot, sequencer))
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/ieee488_2/ieee488_2.py", line 130, in _read_bin
    return self._bin_block_read(flush_line_end)
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/ieee488_2/ieee488_2.py", line 181, in _bin_block_read
    raise RuntimeError(s)
RuntimeError: Header error: received b'0\r'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/scpi/cluster.py", line 54, in decorator_wrapper
    args[0]._check_error_queue(err)
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/scpi/cluster.py", line 5316, in _check_error_queue
    while int(self._read('SYSTem:ERRor:COUNt?')) != 0:
ValueError: invalid literal for int() with base 10: ''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
...
  File "/home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages/qblox_instruments/scpi/cluster.py", line 5316, in _check_error_queue
    while int(self._read('SYSTem:ERRor:COUNt?')) != 0:
ValueError: ('invalid literal for int() with base 10: \'#41067{"awg":[{"cont_mode":{"en_path":[false,false],"wave_idx_path":[0,0]},"gain_path":[1.0,1.0],"marker_ovr":{"en":true,"val":15},"mixer":{"corr_gain_ratio":1.0,"corr_phase_offset_degree":-0.0,"en":', 'setting qblox_cluster_cluster_controller_0_module13_sequencer0_sync_en to True')
```

### System Information

```Shell
Name: qililab
Version: 0.21.1
Summary: Fundamental package for fast characterization and calibration of quantum chips.
Home-page: https://github.com/qilimanjaro-tech/qililab
Author: Qilimanjaro Quantum Tech
Author-email: info@qilimanjaro.tech
License: Apache License 2.0
Location: /home/gtroncoso/miniconda3/envs/qcodes/lib/python3.10/site-packages
Editable project location: /home/gtroncoso/python/qililab
Requires: h5py, lmfit, networkx, pandas, papermill, PyVISA-py, PyYAML, qblox-instruments, qcodes, qcodes-contrib-drivers, qibo, qiboconnection, qm-qua, qpysequence, qualang-tools, ruamel.yaml, rustworkx, submitit, tqdm, urllib3
Required-by: 

Platform info:             Linux-5.15.0-89-generic-x86_64-with-glibc2.35
Python version:            3.10.13
PyVISA version:            0.5.2
QCodes version:            0.38.1
QCodes Contrib version:    0.18.0
Qblox Instrument version:  0.10.1
Qpysequence version:       0.10.0
Quantum Machines version:  1.1.3
Qibo version:              0.1.12.dev0
```

### Existing GitHub issues

- [X] I have searched existing GitHub issues to make sure the issue does not already exist.

visagim, 2023-12-11:
Hi,
> If I reboot the platform, I wil loose all the parameters as they are now, which will mess with the experiment, this cannot happen.
By this do you mean rebooting qiliab's `platform` or are your referring to restarting your notebook's kernel (experimental platform)?
If it is the later, running `platform.disconnect()` and then running the usual instrument setup steps:
```python
platform.connect()
platform.initial_setup()
platform.turn_on_instruments()
```
will reset the cluster and clear its cache, and this should solve the issue.

TronGon, 2023-12-11:
I know this will solve the problem, but if I run platform.connect(), it may crash the yokogawa. Also, it will reset all the parameters to the parameters in the runcard, which may not be the ones at the experiment at that moment.

AlbertMitjans, 2023-12-12:
> I know this will solve the problem, but if I run platform.connect(), it may crash the yokogawa. Also, it will reset all the parameters to the parameters in the runcard, which may not be the ones at the experiment at that moment.

The resetting of the parameters is done inside the `platform.initial_setup()`, thus doing `platform.connect()` shouldn't affect the current instrument parameters.

@visagim please correct me if I'm wrong.

visagim, 2023-12-12:
Running `connect` also initializes the device however since it runs:
```python
    def _initialize_device_and_set_to_all_modules(self):
        """Initialize the Controller Device driver and sets it for all modules"""
        self._initialize_device()
        self._set_device_to_all_modules()
```
This might have unwanted effects on the source mode parameter of the yokogawa. If the only problem is qblox's cluster, I think the best approach would be to get the cluster using `instrument = platform.get_element("instrument_alias")` and then running `instrument.reset()`

jjmartinezQT, 2023-12-20:
This is not a real bug but a misuse. Qililab does not offer the possibility of stopping experiments midway and keep parameters at the moment, which would be a new feature if needed. There are ways to recover all instruments in the event an experiment crashes midway or is stopped, as described in the comments.

AlbertMitjans, 2023-12-20:
@jjmartinezQT I agree with you that this is not a bug. What do you think of keeping this issue as an enhancement?

I will reopen the issue and change the label. Let me know if you don't agree! 😄

jjmartinezQT, 2023-12-20:
@AlbertMitjans Yes, that is fine, we can implement this as an enhancement.